pyusb/pyusb #449: raising correct errors

doronz88, 2022-08-21:
PyUSB should raise more accurate errors, such as `USBBrokenPipeError` instead of `USBError` for everything so users can treat these exceptions better.
I can create this as a PR if you aggree

jonasmalacofilho, 2022-08-22:
PyUSB could indeed benefit from higher level and more (immediately) informative errors, but that's needs to go beyond replacing

```py
raise USBError(..., LIBUSB_ERROR_PIPE, EPIPE)
```

with

```py
raise USBBrokenPipeError()
```

Most errors that a backend may return depend not only on a situation, but also on the backend, platform and kernel/system driver in use.  Not to mention that there are at least 3 different libusb-0.1 implementations...

So a through review of common error scenarios, and the many ways in which they can manifest in our backends, would be necessary in order to provide additional higher level errors.

But, despite this caveat, improvements in this area would be very much appreciated.

mcuee, 2022-08-23:
Maybe we can fix libusb-1.0 backend first.

For libusb-0.1, indeed it may be more complicated.
* libusb-compat-0.1
* libusb-win32 for Windows
* legacy libusb-0.1