pyusb/pyusb #465: Confusing documentation

pluto-dev, 2022-11-09:
I find the the documentation on how to install libusb on windows very confusing.

[faq](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-do-i-install-libusb-on-windows)

From the text I can't figure out if I should install libusb myself or if the library is enough to get going.

> On Windows, [`pyocd/libusb-package`_](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id5) is a convenient [[1]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id3) [[2]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id4) way to provide the necessary libusb 1.0 DLL

The [`pyocd/libusb-package`_](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id5) is just referencing the faq itself

[[1]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id3) [[2]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id4)  is linking to the following in the faq

> Linux and BSD are typically set up in a way in which root permissions are needed for low level/generic access to USB devices.

This is confusing, when the documentation speaks about a Windows issue but linking to an unrelevante Linux problem

jonasmalacofilho, 2022-11-10:
> The [`pyocd/libusb-package`_](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id5) is just referencing the faq itself

That link is indeed broken. I'll fix it ASAP.

> [[1]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id3) [[2]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#id4) is linking to the following in the faq [...]

That's not accurate, they both point to the corresponding footnotes.

jonasmalacofilho, 2022-11-10:
Fixed the link, but it seems that GitHub rendering of reStructuredText is broken right now: clicking the correct link results in an error, instead of opening the libusb-package project page (which exists).

[image]

Hopefully this will get sorted out by itself, soon. Else, I'll try to remember to report it to GitHub (but feel free to do it before me, if you can).

commit 40bab772 (2022-11-10): faq: fix link to pyocd/libusb-package

Fixes: #465
closed 2022-11-10 by jonasmalacofilho