pyusb/pyusb #487: PyUSB does not seem to be properly installed one one computer

calugaruful-dotcom, 2023-07-13:
I have a python script I am running in pycharm Windows 10 and recently I am getting this error: ValueError: PyUSB does not seem to be properly installed.

I have ran this script on this computer before, and it worked. I usually install, libusb, pyvisa, pyvisa-py, pyusb packages and that seems to take care of the backend. For example, if I copy this same script and install the same packages on a different computer, it works and I get no error.

On the non working computer when I run pyvisa-info I can also see the usb backend is not configured:
`PyVISA Version: 1.13.0

Backends:
ivi:
Version: 1.13.0 (bundled with PyVISA)
Binary library: Not found
py:
Version: 0.7.0
TCPIP INSTR: Available
Resource discovery:
- VXI-11: ok
- hislip: ok
TCPIP SOCKET: Available
ASRL INSTR:
Please install PySerial (>=3.0) to use this resource type.
No module named 'serial'
USB INSTR:
PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,
libusb-win32 or OpenUSB.
No backend available
USB RAW:
PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,
libusb-win32 or OpenUSB.
No backend available
VICP INSTR:
Please install PyVICP to use this resource type.
GPIB INSTR:
Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
No module named 'gpib'
GPIB INTFC:
Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
No module named 'gpib'`

`usb.core.find()` gets me this:
2023-07-13 11:23:59,582 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2023-07-13 11:23:59,584 ERROR:usb.libloader:'Libusb 1' could not be found
2023-07-13 11:23:59,584 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2023-07-13 11:23:59,584 ERROR:usb.libloader:'OpenUSB library' could not be found
2023-07-13 11:23:59,584 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2023-07-13 11:23:59,585 ERROR:usb.libloader:'Libusb 0' could not be found
2023-07-13 11:23:59,585 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend

Where is usb find automatically looking for the libraries? I thought it was system32, but adding .dlls there didnt help.

jonasmalacofilho, 2023-07-13:
> Where is usb find automatically looking for the libraries? I thought it was system32, but adding .dlls there didnt help.

That's correct. Although, technically, we delegate that to ctypes, which delegates to the OS (this can sometimes be relevant).

My first suggestion is for you to check that the DLL architecture matches the Python interpreter you're running on. 

After that, use ProcessMonitor to check which paths are being tested, which files are found, etc... maybe something else is overriding the default search paths for DLLs.

closed 2023-07-14 by jonasmalacofilho

calugaruful-dotcom, 2023-07-14:
I wouldnt think the interpreter is the issue, since on the second computer I am using same python version, same interpreter. As for the process monitor suggestion, could you elaborate on that please? Could you give an example how I might I go about doing that?

jonasmalacofilho, 2023-07-15:
> I wouldnt think the interpreter is the issue, since on the second computer I am using same python version, same interpreter.

Personally, I would check it anyway: you're already looking into a "weird" behavior, in that it works on all but one machine. At that point I think any expectations involving the word "same" are out the window... Also, the interpreter _and_ DLLs need to be checked.

> As for the process monitor suggestion, could you elaborate on that please? Could you give an example how I might I go about doing that?

As far as I remember it has a fairly intuitive interface (I don't have a Windows box at hand right now). Just open it and (IIRC) it starts to show the activity from all processes (you can also filter by process name, PID, etc).

Next, start your program—or, preferably, a minimal example that only tries to list all USB devices... `import usb; usb.core.find(find_all=True)`—and watch Process Monitor for what paths are accessed in one way or another.

---

One more thing: I'm not sure how Windows would behave if it finds a DLL with the correct name _but wrong architecture_  in the search path. Does it keep looking for other DLLs (with that name) in (other locations in) the search path anyway, or will it only attempt to load that first, wrong, one?