pyusb/pyusb #489: `bytearray()` buffers in addition to `array.array`

tannewt, 2023-07-25:
Any objection to a PR to support bytearray in addition to array.array buffers? `bytearray()` has the advantage that you can specific the length when you create it instead of needing an initializer object.

Context is this commit which added the implicit readinto support: https://github.com/pyusb/pyusb/commit/1028dc6d213d027d0f1b0b9dceb7157a4bbff47d

jonasmalacofilho, 2024-07-10:
Some methods already support this (those using `_interop.as_array`, e.g. `Device.write`).

Others currently don't support it (e.g. `Device.read`), and I'm not sure it's practical to change them: in the case of `Device.read`, internally we convert the buffer to `array('B')`; therefore we would need to make sure that the conversion and back are zero-copy, and actually remember to convert it back to a `bytearray` before returning, or if any exception is raised...

tannewt, 2024-08-20:
I think `memoryview()` would help in the conversion. (It'd also make the "returns less than requested" case zero copy.)

You could invert the test to create a buffer and only do it when given an int. Anything else you'd want to treat as something that implements the buffer protocol.

jonasmalacofilho, 2024-08-21:
Can you send a draft PR?

Please note that I don't intend to introduce breaking changes to this API. There's still some leeway in what can be defined as an acceptable change (since every visible side-effect of the current API definition and implementation is probably depended on _by someone),_ but at the very least I don't intend to make programs that are unquestionably correct today break tomorrow.