pyusb/pyusb #490: There was an error in the subpackage sending, the first normal sending and the second direct error report, the agreed code c is fine

5tiaowu, 2023-07-27:
[image]
int send_hid_descriptor(libusb_device_handle *handle,
                        const unsigned char *descriptor, uint16_t size,
                        uint8_t max_packet_size_0) {
    const uint8_t requestType = LIBUSB_ENDPOINT_OUT | LIBUSB_REQUEST_TYPE_VENDOR;
    const uint8_t request = AOA_SET_HID_REPORT_DESC;
    const uint16_t value = 0;
    unsigned char *const buffer = (unsigned char *)descriptor;
    const unsigned int timeout = DEFAULT_TIMEOUT;

    uint16_t offset = 0;
    printf("返回值1：%d\n", size);
    printf("返回值2：%d\n", max_packet_size_0);

    // Create a string to store the buffer + offset values
    char str1[3 * size]; // 3 times size to accommodate "xx " for each element
    str1[0] = '\0'; // Initialize the string as an empty string

    while (offset < size) {
        uint16_t packet_length = size - offset;
        printf("size返回值1：%d\n", size);
        printf("offset返回值1：%d\n", offset);
        printf("packet_length返回值1：%d\n", packet_length);
        if (packet_length > max_packet_size_0) {
            packet_length = max_packet_size_0;
        }

        for (uint16_t i = 0; i < packet_length; i++) {
            char temp[4]; // Buffer to store the formatted hex value and space
            sprintf(temp, "%02x ", buffer[offset + i]);
            strcat(str1, temp); // Append the formatted value to the string
        }

        int r = libusb_control_transfer(handle, requestType, request, value,
                                        offset, buffer + offset, packet_length,
                                        timeout);
        offset += packet_length;
        printf("Value at buffer + offset: %02x\n", buffer[offset]);
        printf("返回值3：%d\n", r);
        if (r < 0) {
            print_libusb_error(r);
            return 1;
        }
        printf("buffer + offset: %s\n", str1);
    }

    // Print the entire string containing buffer + offset values

    return 0;
}def send_hid_descriptor(handle, descriptor, size, max_packet_size_0):
    offset = 0
    while offset < size:
        packet_length = size - offset
        if packet_length > max_packet_size_0:
              packet_length = max_packet_size_0
        else:
            print("最后一组")
            # offset = 0
            # packet_length=size
        # packet_length = min(size - offset, max_packet_size_0)
        print(" ".join([f"{byte:02X}" for byte in descriptor[offset:offset + packet_length]]))
        r = handle.ctrl_transfer(
            usb.util.CTRL_OUT | usb.util.CTRL_TYPE_VENDOR,
            ACCESSORY_SET_HID_REPORT_DESC,
            1,
            0,
            descriptor[offset:offset + packet_length],
            DEFAULT_TIMEOUT
        )
        offset += packet_length

        if r < 0:
            print("发送hiddesc失败")
        else:
            print("发送hiddesc成功：" + str(r))The same logic can subcontract normally in c, but not in pysub

5tiaowu, 2023-07-27:
REPORT_DESC_TOUCH_SCREEN = [
    0x05, 0x0d,  # USAGE_PAGE (Digitizers)
    0x09, 0x04,  # USAGE (Touch Screen)
    0xa1, 0x01,  # COLLECTION (Application)
    0x09, 0x22,  # USAGE (Finger)
    0xa1, 0x00,  # COLLECTION (Physical)
    0x09, 0x42,  # USAGE (Tip Switch)
    0x15, 0x00,  # LOGICAL_MINIMUM (0)
    0x25, 0x01,  # LOGICAL_MAXIMUM (1)
    0x75, 0x01,  # REPORT_SIZE (1)
    0x95, 0x01,  # REPORT_COUNT (1)
    0x81, 0x02,  # INPUT (Data,Var,Abs)
    0x09, 0x32,
    0x15, 0x00,
    0x25, 0x01,
    0x81, 0x02,
    0x09, 0x51,
    0x75, 0x05,
    0x95, 0x01,
    0x16, 0x00, 0x00,
    0x26, 0x10, 0x00,
    0x81, 0x02,
    0x09, 0x47,
    0x75, 0x01,
    0x95, 0x01,
    0x15, 0x00,
    0x25, 0x01,
    0x81, 0x02,
    0x05, 0x01,
    0x09, 0x30,
    0x75, 0x10,
    0x95, 0x01,
    0x55, 0x0D,
    0x65, 0x33,
    0x35, 0x00,
    0x46, 0x60, 0x17,
    0x26, 0xFF, 0x0F,
    0x81, 0x02,
    0x09, 0x31,
    0x75, 0x10,
    0x95, 0x01,
    0x55, 0x0D,
    0x65, 0x33,
    0x35, 0x00,
    0x46, 0x26, 0x0E,
    0x26, 0xFF, 0x0F,
    0x81, 0x02,
    0x05, 0x0D,
    0x09, 0x55,
    0x25, 0x08,
    0x75, 0x08,
    0x95, 0x01,
    0xB1, 0x02,
    0xC0,
    0xC0,
]

5tiaowu, 2023-07-27:
register_hid(dev, len(REPORT_DESC_TOUCH_SCREEN))
    send_hid_descriptor(dev, REPORT_DESC_TOUCH_SCREEN, len(REPORT_DESC_TOUCH_SCREEN), max_packet_size_0)

5tiaowu, 2023-07-27:
[image]

closed 2023-07-27 by jonasmalacofilho

mcuee, 2023-09-24:
BTW, it is better to use HIDAPI/cython-hidapi for HID device, not libusb/pyusb.