pyusb/pyusb #504: Very laggy when reading data from device

c0debreaker, 2024-03-18:
I converted a NodeJS code. The NodeJS program is very fluid and I don't see any lagging when I am reading from the device. Here's the NodeJS code https://github.com/fpv-wtf/voc-poc/blob/master/index.js

And here is my code. When I am using this reading from the same device, it is very laggy. What am I doing wrong?
```
#!/usr/bin/env python

import usb.core
import usb.util
import sys
import signal

# find our device
usb_device = usb.core.find(idVendor=0x2ca3, idProduct=0x1f)

if usb_device is None:
    raise ValueError('Device not found!')

cfg = usb_device.get_active_configuration()
intf = cfg[(3,0)]

out_endpoint = usb_device[0][(3,0)][0]
in_endpoint = usb_device[0][(3,0)][1]

# Code copied from https://stackoverflow.com/questions/67453535/i-keep-getting-usb-core-usberror-errno-16-resource-busy-when-trying-to-read
if usb_device.is_kernel_driver_active(0):
    try:
        usb_device.detach_kernel_driver(0)
    except usb.core.USBError as e:
        sys.exit("Could not detatch kernel driver from interface({0}): {1}".format(intf, str(e)))

magic = bytes.fromhex("524d5654")
out_endpoint.write(magic)

def signal_handler(sig, frame):
    # print('You pressed Ctrl+C!')
    usb.util.release_interface(usb_device, intf)
    sys.exit(0)

signal.signal(signal.SIGINT, signal_handler)

while True:
    try:
        data = usb_device.read(in_endpoint.bEndpointAddress, in_endpoint.wMaxPacketSize)
        sys.stdout.buffer.write(data)

    except usb.core.USBError as e:
        data = None
        if e.args == ('Operation timed out',):
            continue
```

Mickey70, 2024-03-30:
Have you had any success solving your problem? I'm struggling with a stubborn time-out problem when I use the script you have provided above. Maybe unrelated, but I do not know.

mcuee, 2024-03-31:
> `data = usb_device.read(in_endpoint.bEndpointAddress, in_endpoint.wMaxPacketSize)`

You should use large amount of data and not one packet by packet.