pyusb/pyusb #517: usb.core.USBError: [Errno 12] Insufficient memory

KevinXuxuxu, 2024-06-02:
Hi there, I'm getting `usb.core.USBError: [Errno 12] Insufficient memory` when writing data to device. My code look something like this:
```
with open(args.kernel, "rb") as f:
    kernel = f.read()
...
print("Loading kernel...")
kernel_size = len(kernel)
dev.ctrl_transfer(0x21, 2, 0, 0, 0)
dev.ctrl_transfer(0x21, 1, 0, 0, struct.pack('I', kernel_size))

dev.write(2, kernel, 100000000)
print("Kernel loaded successfully.")
```
And the stack trace
```
Traceback (most recent call last):
  File "/home/fzxu/Documents/code/linux-sandcastle/load_linux.py", line 74, in <module>
    dev.write(2, kernel, 100000000)
  File "/home/fzxu/anaconda3/lib/python3.9/site-packages/usb/core.py", line 989, in write
    return fn(
  File "/home/fzxu/anaconda3/lib/python3.9/site-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/home/fzxu/anaconda3/lib/python3.9/site-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/home/fzxu/anaconda3/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 12] Insufficient memory
```
I'm making sure that the size set (`100000000`) is larger than the actual data being transfered (around 35mb) but that doesn't seem to work. I'm fairly new to working with pyusb and any further context or information about the meaning or possible reason for this error is appreciated. Thanks!

jonasmalacofilho, 2024-06-02:
@mcuee , it isn't documented what may cause `libusb_bulk_transfer` to return `LIBUSB_ERROR_NO_MEM` ([AFAIK](https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#ga2f90957ccc1285475ae96ad2ceb1f58c)).

Do you have any insights?

mcuee, 2024-06-02:
> @mcuee , it isn't documented what may cause `libusb_bulk_transfer` to return `LIBUSB_ERROR_NO_MEM` ([AFAIK](https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#ga2f90957ccc1285475ae96ad2ceb1f58c)).
> 
> Do you have any insights?

One possibility is here.
https://github.com/libusb/libusb/blob/e678b3fad58a508cbd0a6e6dc777fb48346f948b/libusb/sync.c#L115C2-L123C3

mcuee, 2024-06-02:
@KevinXuxuxu 

Is this an embedded Linux system? Please tell us more about your system (Linux distro and kernel version used) and libusb version used. Thanks.
https://github.com/libusb/libusb/wiki/Troubleshooting

KevinXuxuxu, 2024-06-02:
Thanks for the insights!
The system running this script is Ubuntu 22.04 on a PC (Core i7 + 32G RAM) so I wouldn't consider that as an embedded environment. I'm trying to transfer data of a linux kernel to [PongoOS](https://github.com/checkra1n/PongoOS) running on Iphone X (A11). Not sure if the "insufficient memory" refers to the host OS or the target device.

jonasmalacofilho, 2024-07-14:
What's the value of `kernel_size`?

danielaird27, 2025-05-01:
I just had this issue. I fixed it by chunking the writing process with a chunk size of 1048576 bytes instead of doing one massive write. The chunk size may be arbitrary, that is just the chunk size that worked for me.

DBlauhut, 2026-08-25:
your failed transfer of 100000000 bytes could be related to the kernel module parameter /sys/module/usbcore/parameters/usbfs_memory_mb which must be bigger than your transfersize.