pyusb/pyusb #521: [libusbK]usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed

Flykoki, 2024-08-14:
Hello,
        Try to use pyusb communicate between Android TV and PC. For my PC pyusb working on windows10 with libusbK Usb , and the USB device is Android TV communicate with pyusb via USB cabel .

        Code snapshot : 
       dev = usb.core.find(idVendor = ACCESSORY_VID);
       version = dev.ctrl_transfer(
            usb.util.CTRL_TYPE_VENDOR | usb.util.CTRL_OUT,
                usb.util.CTRL_TYPE_VENDOR | usb.util.CTRL_IN,
                51, 0, 0, 2)

        The error happend when we call "dev.ctrl_transfer()" , Log snapshot :
  File "C:\Users\xxx\Desktop\python_scripts\accessory.py", line 198, in accessory
    version = dev.ctrl_transfer(
              ^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\program\python3.12.2\Lib\site-packages\usb\core.py", line 1082, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\program\python3.12.2\Lib\site-packages\usb\backend\libusb0.py", line 601, in ctrl_transfer
    return _check(_lib.usb_control_msg(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\program\python3.12.2\Lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xc1\xac\xb5\
xbd\xcf\xb5\xcd\xb3\xc9\xcf\xb5\xc4\xc9\xe8\xb1\xb8\xc3\xbb\xd3\xd0\xb7\xa2\xbb\xd3\xd7\xf7\xd3\xc3\xa1\xa3\r\n\n'

jonasmalacofilho, 2024-08-14:
Sorry, but this doesn't appear to be related to PyUSB: it looks more like you got the protocol wrong, your device is misbehaving, or perhaps there's some error at the backend (libusb0, in this case) or OS levels.

---

I don't think it will help, but I noticed someone else reported the same error message on MTK-bypass/bypass_utility#14. And there seem to be a few other hits on Chinese forums, but I don't read the language.

closed 2024-08-14 by jonasmalacofilho

mcuee, 2024-09-20:
Try using libusbK or WinUSB to see if that helps.\
Or upgrade to latest libusb-win32 version. libusb-win32 old version may have some issues with Control Transfer.

Reference:
* https://github.com/pyusb/pyusb/issues/343

cuellarca, 2025-03-23:
Installing Zadig and with libusbK (v3.1.0.0) driver worked for me.