pyusb/pyusb #532: No backend available error alpine linux

NerosKi, 2025-02-24:
I have installed libusb but still get:
```
>>> import usb.backend.libusb1 as libusb1
>>> backend = libusb1.get_backend(find_library=lambda lib: "/usr/lib/libusb-1.0.so.0")
>>> print("Backend loaded:", backend)
Backend loaded: None
```

Loading with ctypes works:
```
>>> import ctypes
>>> ctypes.CDLL("/usr/lib/libusb-1.0.so.0")
<CDLL '/usr/lib/libusb-1.0.so.0', handle 78c50cfae7f0 at 0x78c50d2a4a50>
```

Also:
```
nm -D /usr/lib/libusb-1.0.so.0 | grep libusb_init
0000000000005335 T libusb_init
```

jonasmalacofilho, 2025-02-24:
According to that debug log, LibUSB is failing to initialize – the `libusb_init` call – with `LIBUSB_ERROR_OTHER`.

NerosKi, 2025-02-24:
What could cause that? I am trying to set this up inside a InvenTree docker environment. I installed everything in that environment. Should I also install something on the server itself?

jonasmalacofilho, 2025-03-01:
I would check the LibUSB sources to see which code paths could result in this error.

With containers, lack of privileges could be an issue, but it would be weird to not even be able to *initialize* the library due that.

mcuee, 2025-03-23:
It is quite possible due to container issue.

libusb FAQ: need to make sure usbfs is present.
https://github.com/libusb/libusb/wiki/FAQ#user-content-libusb_init_return_99_LIBUSB_ERROR_OTHER_how_to_troubleshoot_this_one

mcuee, 2026-03-12:
This recently merged PR may help.
* https://github.com/libusb/libusb/pull/1756

@NerosKi 
Would you please try the latest libusb git? Thanks.