pyusb/pyusb #537: Using PyUSB on Alpine Linux 3.22 on Docker

BinarSkugga, 2025-08-13:
Not really an issue but a sort of informative/documented process for using PyUSB in an Alpine docker image. The context is the following: we use raspberry pis and need to communicate with both serial and USB devices that are plugged in. At first we were using a Debian based image but it started weighting down on our clients' networks so we needed switch to Alpine.

This was not straight forward so here's a little breakdown of the core issue. It seems PyUSB doesn't detect libusb correctly when only musl is available. It usually results in this error:

```python
Traceback (most recent call last):
  File "<string>", line 1, in <module>
    import usb; usb.core.find()
                ~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/usb/core.py", line 1321, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
``` 

Users need to install GCC and copy over the built artefacts from `/usr/lib` to `/lib`. Here's the minimum working Dockerfile that I've come up with:

```dockerfile
FROM python:3.13-alpine AS venv-builder

ENV LD_LIBRARY_PATH=/lib:/usr/lib

WORKDIR /work
RUN apk update && \
  apk add --no-cache gcc binutils libusb-dev py3-libusb1 udev
RUN cp /usr/lib/libusb-1.0* /lib/

RUN python -m pip install pyusb

CMD python -c "import usb; print(usb.core.find())"
```

For this build to work, you will have to mount your host machine usb bus and its udev control also:

```bash
docker build -f Dockerfile.test -t test .
docker run -v /dev/bus/usb:/dev/bus/usb -v /run/udev/control:/run/udev/control test
```

Hope this helps anyone coming across this issue.

BinarSkugga, 2025-08-13:
> That said, in my testing just GCC, `libusb-dev`[1](#user-content-fn-1-1209e6472650818365fc16549a104e77) and setting `LD_LIBRARY_PATH` was enough. Copying `/usr/lib/libusb-1.0*` to `/lib` was not necessary, nor installing `py3-libusb1` or `[e]udev`.

Absolutely, I might've forgot to remove some things from my original image. The alias is also unneeded. Could never find that issue with musl from searching with PyUSB/LibUSB keywords. Hope this issue bridge the gap !

jonasmalacofilho, 2025-08-13:
I'll add it to the FAQ.

Thanks!

jonasmalacofilho, 2025-08-17:
Thanks again!

(Made a slight change to the FAQ PR before merging it: switched to Alpine-packaged PyUSB, since that's generally better than installing from pip on top of whatever might already be on the image).

closed 2025-08-17 by jonasmalacofilho