pyusb/pyusb #539: New backend for FTDI serial devices?

nylnx, 2025-08-29:
Hi,
I'm currently working on serial devices that are mostly FTDI-based serial USB adapters. I found out about [pyftdi](https://github.com/eblot/pyftdi) and this would be perfect for my use case, but the Windows support is lacking due to the driver situation on the FTDI side. Now there is also the [pyftdiwin](https://github.com/mariusgreuel/pyftdiwin) fork for better Windows support that actually implements a new backend for the FTDI devices and uses this to talk to the FTDI driver directly.

Somebody already asked in https://github.com/eblot/pyftdi/issues/388, if the code from pyftdiwin could be merged into pyftdi and they didn't want to do it, because they want to stay independent of dependencies, especially OS-specific ones, which is understandable, and they suggested it should be merged into pyusb instead.

Now I see two options on how to integrate it:

- Merge a new backend into pyusb and then add a little bit of plumbing into pyftdi
- OR make pyftdi capable of supporting custom backends and publish the pyftdiwin backend as a separate package that can be pulled in separately

I see pro and cons for both ways and I wanted to ask, which one you would prefer. Get the first vendor-specific backend added to pyusb that could benefit anyone with FTDI devices, or you say that this is out of scope for this project and the other option is preferable.

Regards, nylnx

mcuee, 2025-09-08:
There may be potential licensing issues with pyftdiwin approach. I understand different people may have different opinion in this one.

Last time there was a long discussion for OpenOCD with regard to this FTDI vendor driver issue which resulted in the leaving of a few developers, including the original developer.

mcuee, 2025-09-08:
My opinion is that this is out of scope for pyusb project. But @jonasmalacofilho has the final say.

nylnx, 2025-09-09:
The FTDI backend is essentially a wrapper layer for the FTDI driver which translates USB commands into FTDI driver calls. It's not a fully functional USB backend which works on all devices, but it would support FTDI devices as if they were USB devices on all platforms with native FTDI drivers, which includes Windows, without the messy libusb situation.

This use case is arguably quite weird... To translate a USB protocol to driver calls, which again translate these calls back into the (hopefully) same USB protocol. I didn't expect this to be the preferred solution by pyusb, but I wanted to ask nonetheless, since this would make pyusb a compatibility layer for many similar situations, since native Windows support is still not there.

The only proper solution would probably be https://github.com/pyusb/pyusb/issues/209, but before that it's really difficult to get anything done, at least if you want plug-and-play for Windows.

closed 2025-09-10 by mcuee